vincelwt/RaspberryCast #7: hi,can I cast video on ftp server?

yangxuan8282, 2015-07-15:
it works great,thx ur job,I can wireless cast some online video to TV,it's great
but I have some questions
I have some video files(in MP4 format) on my ftp server
and wanna use the RaspberryCast cast on TV
the Chrome extensions pop up says: success
but the screen,which is connect with RPI was no response

vincent-lwt, 2015-07-15:
I actually never tried to cast from a ftp server. I will try when I have some time and see if it's possible to support it. Let me 2 weeks.

yangxuan8282, 2015-07-15:
@vincent-lwt 
thanks for quick reply
the function now is pretty useful,but if that work will be better
actually there are another issues,the RaspberryCast work on my laptop with Chrome extensions
**Send to Rpi**,**pause**,**stop** all work
but on android,only control work,like **play**,**pause**,but can't cast video
I have two android phone,one is 4.4.4,another is 4.1,same result 
maybe I should put this in a new issues

vincent-lwt, 2015-07-15:
Yes please create a new issue, will check what I can do.

yangxuan8282, 2015-07-15:
@vincent-lwt 
ok,what info should I provided?

vincent-lwt, 2015-07-17:
@yangxuan8282 I tried to cast from a ftp server, but OMXplayer doesn't support ftp streaming.

yangxuan8282, 2015-07-17:
@vincent-lwt ok,should I close this issues?

vincent-lwt, 2015-07-17:
Yes please.

closed 2015-07-18 by yangxuan8282

vincelwt, 2016-11-12:
Bonjour vincent-lwt/RaspberryCast,
Vincent L. vous a laissé un message sur IQ Elite 
Cliquez ici a.iqelite.com/j/QEVgD4/aep/?hg=jAtxUcofqJ pour lire votre message 

L'équipe d'IQ Elite

---------------------------------------------------------------------------
Ce message est destiné à reply@reply.github.com. Si vous ne souhaitez pas recevoir d'autres messages de la part de IQ Elite, veuillez cliquer sur ce lien : a.iqelite.com/common/unsubscribe/?code=05043ba1&email=reply@reply.github.com.&hg=jAtxUcofqJ
IQ Elite est un service de IQ Elite S.L., dont le siège se trouve à : C/ Gran De Gràcia 15, 1a 1a, 08012 Barcelone, Espagne. 
---------------------------------------------------------------------------

vincelwt, 2016-11-12:
Bonjour vincent-lwt/RaspberryCast,
Vincent L. vous a laissé un message sur IQ Elite 
Cliquez ici a.iqelite.com/j/QEVgD4/tvz/?hg=yNZRrPTCtp pour lire votre message 

L'équipe d'IQ Elite

---------------------------------------------------------------------------
Ce message est destiné à reply@reply.github.com. Si vous ne souhaitez pas recevoir d'autres messages de la part de IQ Elite, veuillez cliquer sur ce lien : a.iqelite.com/common/unsubscribe/?code=6d5adad6&email=reply@reply.github.com.&hg=yNZRrPTCtp
IQ Elite est un service de IQ Elite S.L., dont le siège se trouve à : C/ Gran De Gràcia 15, 1a 1a, 08012 Barcelone, Espagne. 
---------------------------------------------------------------------------

vincelwt, 2016-11-22:
Bonjour vincent-lwt/RaspberryCast,

Vincent L. vous a laissé un message sur IQ Elite 

Cliquez ici a.iqelite.com/j/QEVgD4/aep/?hg=6tkJbB3FEO pour lire votre message 

L'équipe d'IQ Elite

---------------------------------------------------------------------------
Ce message est destiné à reply@reply.github.com. Si vous ne souhaitez pas recevoir d'autres messages de la part de IQ Elite, veuillez cliquer sur ce lien : a.iqelite.com/common/unsubscribe/?code=05043ba1&email=reply@reply.github.com.&hg=6tkJbB3FEO
IQ Elite est un service de IQ Elite S.L., dont le siège se trouve à : C/ Gran De Gràcia 15, 1a 1a, 08012 Barcelone, Espagne. 
---------------------------------------------------------------------------

vincelwt, 2016-11-22:
Bonjour vincent-lwt/RaspberryCast,

Vincent L. vous a laissé un message sur IQ Elite 

Cliquez ici a.iqelite.com/j/QEVgD4/tvz/?hg=oCrW1OwHet pour lire votre message 

L'équipe d'IQ Elite

---------------------------------------------------------------------------
Ce message est destiné à reply@reply.github.com. Si vous ne souhaitez pas recevoir d'autres messages de la part de IQ Elite, veuillez cliquer sur ce lien : a.iqelite.com/common/unsubscribe/?code=6d5adad6&email=reply@reply.github.com.&hg=oCrW1OwHet
IQ Elite est un service de IQ Elite S.L., dont le siège se trouve à : C/ Gran De Gràcia 15, 1a 1a, 08012 Barcelone, Espagne. 
---------------------------------------------------------------------------