vincelwt/RaspberryCast #8: can't cast on android

yangxuan8282, 2015-07-15:
**RaspberryCast** work fine on my laptop with Chrome extensions
**Send to Rpi**,**pause**,**stop** all work
but on android,only control function works,like **play**,**pause**,but can't cast video

I have two android phone,one is 4.4.4,another is 4.1,same result

vincent-lwt, 2015-07-15:
How are you trying to cast a video ? Using the share-to method or by copy/pasting a link into the app ? If you used the share method, from which app have you tried ?

yangxuan8282, 2015-07-16:
@vincent-lwt  I have tried both of them,in app **share to** and paste a link
the app is in Chinese,called Bilibili

vincent-lwt, 2015-07-16:
@yangxuan8282 I think the problem come from youtube-dl, which doesn't fully support BiliBili. I tried extracting a video from BiliBili, but I only got a little sample. Furthermore, the videos seems to be encoded in a format OMXPlayer can't decode. Please try with Youtube or Vimeo (if you can) to make sure the problem doesn't come from the app.

yangxuan8282, 2015-07-16:
@vincent-lwt I have tried viemo,it works(since Youtube is blocked here,I can't test it)
as for **Bilibili**,this site is different from Youtube

some video is stored in their own server,this kind of video can be downloaded by **youtube-dl** and **RaspberryCast** Chrome extensions works fine

others is stored in other video sites(such as youku.com, letv.com,tudou.com,these are famous video sites in China,and they are all listed in this page http://rg3.github.io/youtube-dl/supportedsites.html)
this kind of video are supported by **youtube-dl**,but not supported by **RaspberryCast** Chrome extensions

the android app can't cast both of them( both **share to** and **paste** won't work)

vincent-lwt, 2015-07-17:
@yangxuan8282 If I understand, some videos, (the ones hosted in their own server) works with the Chrome extension but not with the Android App ?

yangxuan8282, 2015-07-17:
@vincent-lwt yes

yangxuan8282, 2015-07-17:
@ for example,this one http://www.bilibili.com/video/av2587296/

vincent-lwt, 2015-07-18:
This video doesn't works with the Chrome Extension for me. Is the URL you tried to paste in the Android App exactly the same as the one you used for the Chrome Extension ?

yangxuan8282, 2015-07-18:
yes,I am sure use same URL

vincent-lwt, 2015-07-18:
I think the problem come from the connection of your phone to Raspberry Pi because the mechanism for casting a video is the same for the Chrome Extension and the inside text box of the Android app (which is a reflect of TheRaspberryIp:2020/remote . Please try casting the video from this page on your computer.

yangxuan8282, 2015-07-18:
what do u mean by 'Please try casting the video from this page on your computer.'?
this url? http://www.bilibili.com/video/av2587296/

vincent-lwt, 2015-07-18:
From your computer, please go on the url : http://YourPiAdress:2020/remote
and paste the video URL in the field.

yangxuan8282, 2015-07-18:
I have visited htttp://<myip>:2020/remote from computer,and paste the URL,it works,the video play on RPI
but when I do same thing on android phone browser,it failed,just flash for one second,but then nothing happen

vincent-lwt, 2015-07-18:
> I have visited htttp://:2020/remote from computer,and paste the URL,it

What is your phone and your phone's browser ?

yangxuan8282, 2015-07-19:
Xiaomi mi 3(Chinese company)
Chrome

vincent-lwt, 2015-07-19:
Should works, because the webpage is exactly the same on your computer and on your phone. Please try with another video and if it doesn't works, with another phone.

closed 2015-09-12 by vincent-lwt

vincelwt, 2016-11-12:
Bonjour vincent-lwt/RaspberryCast,
Vincent L. vous a laissé un message sur IQ Elite 
Cliquez ici a.iqelite.com/j/QEVgD4/cbd/?hg=OIe2b7YEDz pour lire votre message 

L'équipe d'IQ Elite

---------------------------------------------------------------------------
Ce message est destiné à reply@reply.github.com. Si vous ne souhaitez pas recevoir d'autres messages de la part de IQ Elite, veuillez cliquer sur ce lien : a.iqelite.com/common/unsubscribe/?code=aee899ae&email=reply@reply.github.com.&hg=OIe2b7YEDz
IQ Elite est un service de IQ Elite S.L., dont le siège se trouve à : C/ Gran De Gràcia 15, 1a 1a, 08012 Barcelone, Espagne. 
---------------------------------------------------------------------------

vincelwt, 2016-11-12:
Bonjour vincent-lwt/RaspberryCast,
Vincent L. vous a laissé un message sur IQ Elite 
Cliquez ici a.iqelite.com/j/QEVgD4/1cd/?hg=mrukH7xYXG pour lire votre message 

L'équipe d'IQ Elite

---------------------------------------------------------------------------
Ce message est destiné à reply@reply.github.com. Si vous ne souhaitez pas recevoir d'autres messages de la part de IQ Elite, veuillez cliquer sur ce lien : a.iqelite.com/common/unsubscribe/?code=25cd772e&email=reply@reply.github.com.&hg=mrukH7xYXG
IQ Elite est un service de IQ Elite S.L., dont le siège se trouve à : C/ Gran De Gràcia 15, 1a 1a, 08012 Barcelone, Espagne. 
---------------------------------------------------------------------------

vincelwt, 2016-11-22:
Bonjour vincent-lwt/RaspberryCast,

Vincent L. vous a laissé un message sur IQ Elite 

Cliquez ici a.iqelite.com/j/QEVgD4/cbd/?hg=koa3LHGPm8 pour lire votre message 

L'équipe d'IQ Elite

---------------------------------------------------------------------------
Ce message est destiné à reply@reply.github.com. Si vous ne souhaitez pas recevoir d'autres messages de la part de IQ Elite, veuillez cliquer sur ce lien : a.iqelite.com/common/unsubscribe/?code=aee899ae&email=reply@reply.github.com.&hg=koa3LHGPm8
IQ Elite est un service de IQ Elite S.L., dont le siège se trouve à : C/ Gran De Gràcia 15, 1a 1a, 08012 Barcelone, Espagne. 
---------------------------------------------------------------------------

vincelwt, 2016-11-22:
Bonjour vincent-lwt/RaspberryCast,

Vincent L. vous a laissé un message sur IQ Elite 

Cliquez ici a.iqelite.com/j/QEVgD4/1cd/?hg=PlUTi2zqNY pour lire votre message 

L'équipe d'IQ Elite

---------------------------------------------------------------------------
Ce message est destiné à reply@reply.github.com. Si vous ne souhaitez pas recevoir d'autres messages de la part de IQ Elite, veuillez cliquer sur ce lien : a.iqelite.com/common/unsubscribe/?code=25cd772e&email=reply@reply.github.com.&hg=PlUTi2zqNY
IQ Elite est un service de IQ Elite S.L., dont le siège se trouve à : C/ Gran De Gràcia 15, 1a 1a, 08012 Barcelone, Espagne. 
---------------------------------------------------------------------------