vincelwt/RaspberryCast #9: Suggestion: cast a playlist

buzzbo, 2015-07-26:
This is not an issue per se, but I was wondering if it might be possible to cast a youtube playlist. Thoughts?

LoicLWRT, 2015-07-26:
Indeed, it is not possible at the moment to cast a playlist, that would be a great feature and quite easy to implement. 
Unfortunately the current playlist management is fairly limited, we'd have to improve that beforehand.
As for the current implementation, I believe RaspberryCast will only cast the video present in the URL.

vincelwt, 2015-12-17:
Glad to inform it is now possible to cast a playlist :+1: 
Works with Youtube and Soundcloud as of today.

I recommend doing a fresh install, because we made quite a lot of modifications.

closed 2015-12-17 by vincelwt