vincelwt/RaspberryCast #11: windows cast

dynamitemedia, 2016-01-08:
if i wanted to create a app to connect to the pi from my desktop what do you suggest and what protocol do i need to send it to your app?

this looks great thanks!

vincelwt, 2016-01-08:
Hi :) !

You can make basic HTTP requests to control RaspberryCast.

Here are the some of the possible commands :

In those two commands, the URL to cast needs to be Url encoded.

/stream?url=<TheUrlToCast>
/queue?url=<TheUrlToCast>
/video?control=pause
/video?control=stop
/video?control=right
/video?control=left
/sound?vol=more
/sound?vol=less

Here is an example:

http://raspberrycast.local:2020/stream=https%3A%2F%2Fwww.youtube.com%2Fwatch%3Fv%3D0wEcYPSm_f8

However if your system isn't compatible with Bonjour, you need to replace raspberrycast.local by the IP.

For more information you can review the [server code](https://github.com/vincelwt/RaspberryCast/blob/master/server.py).

dynamitemedia, 2016-01-08:
great so i can send over streaming videos too perfect... now do we have to install anything server like ginx or anything or this is it?

vincelwt, 2016-01-09:
If you want to send a video from an online server or a service like Youtube, no need for an additional server.

However, if you want to send a file stored on your computer, you will need to serve the file.

Note that RaspberryCast will automatically convert "localhost" and "127.0.0.1" address to the corresponding local IP .

A client with a built-in server is [already available](https://github.com/vincelwt/RaspberryCast#drag-and-drop-videos-from-computer) though it as never been tested on Windows. I think it should work without problem once GTK3 is installed on the computer.

dynamitemedia, 2016-01-10:
So is there a certain  type of stream protocol I need to send ? Rtmp or HLS?  And I'm assuming it just passed on to omxplayer .   I'm not seeing in the code how omxplayer grabs it

vincelwt, 2016-01-13:
I think it should work with HLS but not sure about Rtmp. And yup, it is passed to OMXplayer. This part is located in [this file](https://github.com/vincelwt/RaspberryCast/blob/master/process.py).

dynamitemedia, 2016-01-13:
Awesome thanks will check it out, have u tested on the new pi?

vincelwt, 2016-01-14:
If by the new pi you mean the Raspberry Pi 2, yes tested and fully working. It is actually recommended to use it on this one. However I haven't tried it on the new Raspberry Pi Zero.

closed 2016-01-21 by vincelwt