vincelwt/RaspberryCast #12: youtube-dl error

yilmazerhan, 2016-01-20:
How can I solve that problem?

Traceback (most recent call last):
  File "./server.py", line 25, in <module>
    from process import *
  File "/home/pi/RaspberryCast/process.py", line 1, in <module>
    import youtube_dl, os, threading, logging, json
ImportError: No module named youtube_dl
^C

AdityaNG, 2016-01-20:
You most likely don't have youtube-dl installed. A simple install should fix your problem 
```
sudo apt-get install youtube-dl 
```
After running the above code, try to reinstall RaspberryCast and run it again.

vincelwt, 2016-01-20:
What system are you running ?
Did you installed it using the setup.sh method ? This should have installed youtube-dl via pip.

Otherwise, you can try 
>sudo pip install youtube-dl

and reboot.

yilmazerhan, 2016-01-20:
I tried reinstallyoutube-dl and RaspberryCast but coulnd't solve the issue. Then I installed lastest version of OS then tried again. Now it is  working. Thanks

vincelwt, 2016-01-21:
Great :) Enjoy !

closed 2016-01-21 by vincelwt