vincelwt/RaspberryCast #15: Errno 98 Fix?

birchtree02, 2016-03-24:
Every time I activate the Pi, it pops up with the 'Ready To Cast' image and then returns some text, ending with:

`socket.error: [Errno 98] Address already in use`

I have little programming knowledge, but would it be possible to put a 'Try' in and then use another Port if it doesn't work?

vincelwt, 2016-03-24:
Um,  did you try rebooting the Pi?
It usually fix the error.
I imagine you did, so the problem must be with another process using the
2020 port.

Are you using a clean install or did you installed some programs on it?

birchtree02, 2016-03-24:
Yes, I have installed other programs on it. Is there a way to stop them using the 2020 port?
Or, if not, how can I clean install Raspberrycast?

LoicLWRT, 2016-03-24:
Stylée ta réponse

vincelwt, 2016-03-24:
First, you should check which process is listening on 2020 port.
Run the command 
`netstat -tulpn`

If the process is "python" it probably means an instance of RaspberryCast is running.

The problem actually doesn't come from RaspberryCast's code, but from Bottle.py (a server we are using). So using a Try wouldn't work, except if we modify Bottle's code.

You could change the port in server.py file, but then you would have to change the code of the Firefox/Chrome extension accordingly.

Thing is, I don't know any other program using 2020 port (that's why I chose it) so maybe the problem come from the fact you are opening RaspberryCast twice ? Maybe you set up a script at boot to start it ? (there is already one)

However if you want to clean install, I recommend flashing a clean Raspbian image and running again the setup.sh file.

birchtree02, 2016-03-25:
Thank you very much for your help. I ended up flashing raspbian again and then reinstalling. It is working well but it has stopped playing once or twice. Thanks though.

closed 2016-03-25 by birchtree02

vincelwt, 2016-03-25:
Great ! enjoy

vincelwt, 2016-11-12:
Bonjour vincelwt/RaspberryCast,
Vincent L. vous a laissé un message sur IQ Elite 
Cliquez ici a.iqelite.com/j/QEVgD4/92u/?hg=yNZRrPTCtp pour lire votre message 

L'équipe d'IQ Elite

---------------------------------------------------------------------------
Ce message est destiné à reply@reply.github.com. Si vous ne souhaitez pas recevoir d'autres messages de la part de IQ Elite, veuillez cliquer sur ce lien : a.iqelite.com/common/unsubscribe/?code=e1dec8dc&email=reply@reply.github.com.&hg=yNZRrPTCtp
IQ Elite est un service de IQ Elite S.L., dont le siège se trouve à : C/ Gran De Gràcia 15, 1a 1a, 08012 Barcelone, Espagne. 
---------------------------------------------------------------------------

vincelwt, 2016-11-22:
Bonjour vincelwt/RaspberryCast,

Vincent L. vous a laissé un message sur IQ Elite 

Cliquez ici a.iqelite.com/j/QEVgD4/92u/?hg=oCrW1OwHet pour lire votre message 

L'équipe d'IQ Elite

---------------------------------------------------------------------------
Ce message est destiné à reply@reply.github.com. Si vous ne souhaitez pas recevoir d'autres messages de la part de IQ Elite, veuillez cliquer sur ce lien : a.iqelite.com/common/unsubscribe/?code=e1dec8dc&email=reply@reply.github.com.&hg=oCrW1OwHet
IQ Elite est un service de IQ Elite S.L., dont le siège se trouve à : C/ Gran De Gràcia 15, 1a 1a, 08012 Barcelone, Espagne. 
---------------------------------------------------------------------------