vincelwt/RaspberryCast #16: Stop working suddenly.

yangxuan8282, 2016-03-26:
After cast a video, it works at first. But after a while, the video disappear ( black screen, not return to desktop ). And I have test several video from YouTube , but got same result.

OS: RASPBIAN JESSIE ( latest ) 
Model: Raspberry pi 3

vincelwt, 2016-03-26:
Hi,
Could you attach the log ? It is located at /home/pi/RaspberryCast/RaspberryCast.log

yangxuan8282, 2016-03-27:
03-26 13:52:31 - INFO - Server successfully started!
    03-26 13:52:53 - DEBUG - Running state as been asked : 0
    03-26 13:53:31 - DEBUG - Running state as been asked : 0
    03-26 13:53:32 - DEBUG - Received URL to cast: https://www.youtube.com/watch?v=0wEcYPSm_f8
    03-26 13:53:32 - DEBUG - No subtitles for this stream
    03-26 13:53:32 - INFO - Extracting source video URL...
    03-26 13:53:32 - DEBUG - Parsing source url for https://www.youtube.com/watch?v=0wEcYPSm_f8 with subs :False
    03-26 13:53:33 - DEBUG - [youtube] 0wEcYPSm_f8: Downloading webpage
    03-26 13:53:36 - DEBUG - [youtube] 0wEcYPSm_f8: Downloading video info webpage
    03-26 13:53:39 - DEBUG - [youtube] 0wEcYPSm_f8: Extracting video information
    03-26 13:53:39 - DEBUG - [youtube] 0wEcYPSm_f8: Downloading MPD manifest
    03-26 13:53:43 - DEBUG - Youtube link detected, extracting url in 360p
    03-26 13:53:43 - DEBUG - Full video URL fetched.
    03-26 13:53:43 - INFO - Starting OMXPlayer now.
    03-26 13:54:39 - INFO - Playlist empty, skipping.
    03-26 13:55:11 - INFO - Server successfully started!
    03-26 13:55:57 - DEBUG - Remote page requested.
    03-26 13:56:03 - DEBUG - Received URL to cast: https://youtu.be/qALLVvsY5Uo
    03-26 13:56:03 - DEBUG - No subtitles for this stream
    03-26 13:56:03 - INFO - Extracting source video URL...
    03-26 13:56:03 - DEBUG - Parsing source url for https://youtu.be/qALLVvsY5Uo with subs :False
    03-26 13:56:05 - DEBUG - [youtube] qALLVvsY5Uo: Downloading webpage
    03-26 13:56:07 - DEBUG - [youtube] qALLVvsY5Uo: Downloading video info webpage
    03-26 13:56:08 - DEBUG - [youtube] qALLVvsY5Uo: Extracting video information
    03-26 13:56:08 - DEBUG - [youtube] qALLVvsY5Uo: Downloading MPD manifest
    03-26 13:56:10 - DEBUG - Youtube link detected, extracting url in 360p
    03-26 13:56:10 - DEBUG - Full video URL fetched.
    03-26 13:56:10 - INFO - Starting OMXPlayer now.
    03-26 13:56:23 - INFO - REMOTE: Command : Sound --
    03-26 13:56:24 - INFO - REMOTE: Command : Sound --
    03-26 13:56:24 - INFO - REMOTE: Command : Sound --
    03-26 13:56:25 - INFO - REMOTE: Command : Sound --
    03-26 13:56:25 - INFO - REMOTE: Command : Sound --
    03-26 13:56:52 - INFO - REMOTE: Command : Sound ++

yangxuan8282, 2016-03-27:
I do a fresh install raspbian jessie. Now it did finished play video at first, but after that, no video output anymore(still not return to desktop, the RPI3 is connect to an old CRT TV via RCA cable). When I use HDMI output before, never got this problem.

vincelwt, 2016-03-27:
Umm, I think it might come from fbi (the image terminal viewer used to show the RaspberryCast image) which output the image to a specific endpoint. 

Can you still play a video, even after the black screen or does it make the Raspberry Pi unavalaible?

vincelwt, 2016-03-27:
I found something on this page : http://elinux.org/Omxplayer

You can try to use command :
xrefresh -display :0

After running 
sudo apt-get install x11-xserver-utils

And see if the screen comes back to life.

yangxuan8282, 2016-03-27:
@vincelwt after the black screen, the RPI is unavailable via GUI

    xrefresh -display :0

not work (x11-xserver-utils is already installed)

yangxuan8282, 2016-03-27:
I also have tried to give more memory for GPU, but same result

vincelwt, 2016-03-28:
Unfortunately I do not have an old TV around to try :/
Could you try, after playing a video and after the black screen to connect via SSH, and try the following commands:
`cd RaspberryCast`
`sudo ./RaspberryCast.sh stop`

It should stop RaspberryCast and Omxplayer. Maybe it would release your Pi.

yangxuan8282, 2016-03-29:
sudo ./RaspberryCast.sh stop
not work either
I guess I have to pick up a HDMI monitor

BTW, sorry for the issues I submitted last year, after submitted that issues, I move to new flat, don't have monitor to test. But I should closed the issues earlier.

closed 2016-03-29 by yangxuan8282

vincelwt, 2016-03-30:
No worry !

yangxuan8282, 2016-04-08:
@vincelwt I am not sure should report here or submit a new issues. But I got same issues on a HDMI monitor (Motorola Lapdock). After cast first video, the screen stop on the RaspberryCast image.(I guess you are right, this maybe caused by fbi, can I disable it ?) Can't return to desktop or cast next video.  

    sudo ./RaspberryCast.sh stop

not work.It did says kill the service, but the screen not change.Have to reboot. 

Also,I found a strange act of **git clone**:

    [root@localhost ~]# git clone https://github.com/vincelwt/RaspberryCast
    Initialized empty Git repository in /root/RaspberryCast/.git/
    remote: Counting objects: 1762, done.
    remote: Total 1762 (delta 0), reused 0 (delta 0), pack-reused 1762
    Receiving objects: 100% (1762/1762), 88.93 MiB | 10.57 MiB/s, done.
    Resolving deltas: 100% (777/777), done.

The real size of this git should be 1.91MB, but it report 88.93MB. The local git size is 1.91MB, but the git clone did take long time.
I have tested on three computers with different network connection ( my laptop,RPi, my VPS in US), got same result.

vincelwt, 2016-04-08:
Mmh, then the problem must come from somewhere else that the use of CRT.

You can actually disable Fbi to see if it solve the issue. For this, edit **raspberrycast.conf** and replace at "new_log" true by false. Then reboot.

About the git clone:

Thanks for pointing this out !
The problem is that it downloads every commits of the repo.
The solution is to use : 
`git clone https://github.com/vincelwt/RaspberryCast.git --depth=1`
Which will only download the latest commit.

yangxuan8282, 2016-04-09:
@vincelwt I disable the fbi in **raspberrycast.conf**. Then it works, but just a moment. After cast 3 video,it didn't work again(this time, it return to desktop,no need to reboot),I have to restart the service.
Is there any method to use a older version? I realize that the 2015-07 version work pretty well for me.

vincelwt, 2016-04-10:
I can't see where the issue come from :/

Anyway, if you want to use an older version : 
Open the repo on the Pi:
 
> cd RaspberryCast/

(it needs to be the full repo, cloned without --depth=1)

Revert to the commit you want, for example for Dec 3 (before fbi and co)
`git checkout 8fea6befdc149f2ca86f6ed4c63192e0c347d4f7 .`
Do not forget the "." at the end of the command.
This version is also before auto-update so it should stay like that.

You can have the list of commits here : 
https://github.com/vincelwt/RaspberryCast/commits/master

You could try to go back up to 2015-07 but I don't think the Chrome extension would be compatible with these olders versions :/

aflockofseacows, 2017-09-24:
sorry to bring this up again, but I found a fix and I was thinking someone else might have this problem, so I'm gonna put it here.

In the file "process.py", I removed the option "-b" from lines 117 and 121. 
"-b" creates a blank background in omxplayer and I guess, for whatever reason, it never deletes this background and it covers anything else going on.