vincelwt/RaspberryCast #17: Audio card

heretic707, 2016-05-20:
Is there any way to change the audio output besides embedded on the HDMI?

vincelwt, 2016-05-20:
Hi,

By default, the audio output on both HDMI and jack.

If you want to force either HDMI or Sound output, you can edit lines 112 & 116 of **process.py** file, and replace 'both' with 'hdmi' for hdmi output or 'local' for sound output.

heretic707, 2016-05-27:
Thank you for your help!

heretic707, 2016-05-29:
Do you think you can help me? I apologize if this isn't the correct place to ask but.... I have a hifi berry amp installed on my raspberry pi, audio comes out fine, But It would be awesome if I can cast audio out to my screen and have it come out of the amp. I did like you suggested and added "local" but audio still comes out of HDMI. I'm new to linux and python but learning.... Am I doing something wrong?

vincelwt, 2016-05-30:
I think your problem is about the audio driver of your install for the Hifi berry's amp. 
I don't own one so I can't really help you, but you should try to post on Hifiberry's forum where they should be more qualified to answer that issue.

Good luck !

heretic707, 2016-05-30:
Thank you! I appreciate your feedback.

closed 2016-05-30 by vincelwt