vincelwt/RaspberryCast #18: Stuck on "RaspberryCast is ready" screen; unsupported URL in log

shu8, 2016-05-29:
I'm trying to stream a webpage with the RaspberryCast Chrome extension, and I get the notifications saying it is processing, and that it should appear in a few seconds on the screen, but all that happens is I get the 'RaspberryCast is ready' screen and it doesn't change.

Looking at the log, I get:

```
(05-29 17:16:59 - INFO - Server successfully started!
05-29 17:18:10 - DEBUG - Running state as been asked : 0
05-29 17:18:12 - DEBUG - Received URL to cast: https://github.com/lanceseidman/PiCAST
05-29 17:18:12 - DEBUG - No subtitles for this stream
05-29 17:18:12 - INFO - Extracting source video URL...
05-29 17:18:12 - DEBUG - Parsing source url for https://github.com/lanceseidman/PiCAST with subs :False
05-29 17:18:28 - DEBUG - [generic] PiCAST: Requesting header
05-29 17:18:29 - WARNING - Falling back on generic information extractor.
05-29 17:18:29 - DEBUG - [generic] PiCAST: Downloading webpage
05-29 17:18:30 - DEBUG - [generic] PiCAST: Extracting information
05-29 17:18:33 - ERROR - ^[[0;31mERROR:^[[0m Unsupported URL: https://github.com/lanceseidman/PiCAST
05-29 17:18:33 - ERROR - Result is none, returning none. Cancelling following function.
05-29 17:18:33 - DEBUG - Full video URL fetched.
05-29 17:18:33 - INFO - Starting OMXPlayer now.
05-29 17:18:33 - INFO - Playlist empty, skipping.)
```

Why is it unsupported? Are there any specific URLs that are supported?

I'd be happy to provide more logs or anything to help fix this! :)

vincelwt, 2016-05-29:
Hello,

Sorry but as of now RaspberryCast only support media content (audio and
video) ,  and doesn't support casting of websites.

You can cast Youtube pages or media websites but not regular pages.

Vincent L.

shu8, 2016-05-29:
@vincelwt oh! sorry! I thought it could do sites!

Thanks for the help though! This is still a really great program which is going to be really useful for me! :smiley:

closed 2016-05-29 by shu8

vincelwt, 2016-11-12:
Bonjour vincelwt/RaspberryCast,
Vincent L. vous a laissé un message sur IQ Elite 
Cliquez ici a.iqelite.com/j/QEVgD4/p7h/?hg=jAtxUcofqJ pour lire votre message 

L'équipe d'IQ Elite

---------------------------------------------------------------------------
Ce message est destiné à reply@reply.github.com. Si vous ne souhaitez pas recevoir d'autres messages de la part de IQ Elite, veuillez cliquer sur ce lien : a.iqelite.com/common/unsubscribe/?code=1c18aa35&email=reply@reply.github.com.&hg=jAtxUcofqJ
IQ Elite est un service de IQ Elite S.L., dont le siège se trouve à : C/ Gran De Gràcia 15, 1a 1a, 08012 Barcelone, Espagne. 
---------------------------------------------------------------------------

vincelwt, 2016-11-22:
Bonjour vincelwt/RaspberryCast,

Vincent L. vous a laissé un message sur IQ Elite 

Cliquez ici a.iqelite.com/j/QEVgD4/p7h/?hg=6tkJbB3FEO pour lire votre message 

L'équipe d'IQ Elite

---------------------------------------------------------------------------
Ce message est destiné à reply@reply.github.com. Si vous ne souhaitez pas recevoir d'autres messages de la part de IQ Elite, veuillez cliquer sur ce lien : a.iqelite.com/common/unsubscribe/?code=1c18aa35&email=reply@reply.github.com.&hg=6tkJbB3FEO
IQ Elite est un service de IQ Elite S.L., dont le siège se trouve à : C/ Gran De Gràcia 15, 1a 1a, 08012 Barcelone, Espagne. 
---------------------------------------------------------------------------