vincelwt/RaspberryCast #19: All is I see is a RaspberryCast splash screen

Sachin1968, 2016-05-30:
Installed on a Pi 3, I can access the Remote via http or the Android app. I'm also able to add videos to the queue via the Chrome extension but all I see on the Pi screen is a screen that says RaspberryCast is ready.

Any suggestions?

Thanks, I really want this to work!

vincelwt, 2016-05-30:
Hello, 

Could you provide the log, located at _/home/pi/RaspberryCast/RaspberryCast.log_ ?

edberteliezer, 2016-08-20:
same problem. could be related to ip adress? its set to raspberry.local

edberteliezer, 2016-08-20:
here is a log file pastebin.com/9JUc0ErX

edberteliezer, 2016-08-20:
so basically every video i casted returns [0;31mERROR:[0m no suitable InfoExtractor for URL

vincelwt, 2016-08-20:
@edberteliezer Thanks for posting a log.

For the Youtube videos it seems that youtube-dl isn't working (as #22 ). Could you try manually reinstalling it ?

vincelwt, 2016-08-20:
Also, the first videos in the log doesn't seem to come from YouTube, except bug from RaspberryCast end. Does they come from YouTube ? Or from another Google service (like it seems) ?

edberteliezer, 2016-08-20:
I will try reinstalling the youtube-dl. And yes, the first video is from a google service, presumably Google Drive, uploaded and extracted the direct video stream url using, presumably, similar method of youtube-dl -g [URL]. It is actually a.. um.. 'fan reupload' of.. a thing that was on TV :)

vincelwt, 2016-08-20:
Ok, thanks for pointing this out. The problem with these videos is that youtube-dl try to extract the url. I will try to correct this soon

edberteliezer, 2016-08-21:
Ok, so pip is bugged
https://github.com/pypa/pip/issues/2964#issuecomment-137346639
https://github.com/pypa/pip/issues/2964#issuecomment-192607458

I suppose I will uninstall with apt-get

edberteliezer, 2016-08-21:
now both chrome and remote web says please make sure link/action is compatible. I will try removing the RaspberryCast and Youtube-dl and reinstall them is thats a help

edberteliezer, 2016-08-21:
So, after tinkering in the web for a while, I installed youtube-dl with pip --ingnore-installed. and again with --upgrade. Now, I can properly uninstall and install again with pip. 

And the result is ERROR PLEASE MAKE SURE LINK IS COMPATIBLE. 

Log file: http://pastebin.com/V8DC5MZU

vincelwt, 2016-08-21:
Thanks for trying. Your issue the same as #22 . 
I think it may be due to new versions of youtube-dl which changed the output data structure.

Will make some tests this week

@Sachin1968 could you try to reinstall youtube-dl too ?

edberteliezer, 2016-08-22:
Well Good luck then. So far you are the only promising mp4 / youtube-dl url 'omx caster' that is up-to date. Most of my findings are from 2012-2014 which is out dated etc. Also, current on-going 'mirroring' project worth taking a look: https://www.youtube.com/watch?v=yhxnAmJsT_E&feature=youtu.be&list=PLT2HCg4Nbo1upDmqPFxyT8koQwvCcUOph | Maybe you can collaborate for experimental mirroring button or something. The problem with that project is it installs other presentation stuff which I dont need and too many commands to input which will take some times to read and figure out.

mheine, 2018-07-04:
This issue is now almost 2 years old, and as such I'm closing it.

closed 2018-07-04 by mheine