vincelwt/RaspberryCast #20: Some Youtube videos cannot be casted

boskodin, 2016-06-06:
https://www.youtube.com/watch?v=5s7_WbiR79E
for example

vincelwt, 2016-06-06:
Hello, 

Could you provide the log, located at /home/pi/RaspberryCast/RaspberryCast.log ?

boskodin, 2016-06-06:
06-06 11:23:49 - DEBUG - Parsing source url for https://www.youtube.com/watch?v=5s7_WbiR79E with subs :False
06-06 11:23:50 - DEBUG - [youtube] 5s7_WbiR79E: Downloading webpage
06-06 11:23:50 - DEBUG - [youtube] 5s7_WbiR79E: Downloading video info webpage
06-06 11:23:51 - DEBUG - [youtube] 5s7_WbiR79E: Extracting video information
06-06 11:23:51 - DEBUG - [youtube] 5s7_WbiR79E: Downloading MPD manifest
06-06 11:23:57 - DEBUG - CASTING: Youtube link detected, extracting url in maximal quality.
06-06 11:23:57 - DEBUG - Full video URL fetched.
06-06 11:23:57 - INFO - Starting OMXPlayer now.
06-06 11:24:02 - INFO - Starting next video in playlist.
06-06 11:24:02 - INFO - Starting OMXPlayer now.
06-06 11:24:06 - INFO - Playlist empty, skipping.

boskodin, 2016-06-06:
continued, trying to play another one that didn't cast

06-06 11:23:57 - INFO - Starting OMXPlayer now.
06-06 11:24:02 - INFO - Starting next video in playlist.
06-06 11:24:02 - INFO - Starting OMXPlayer now.
06-06 11:24:06 - INFO - Playlist empty, skipping.
06-06 11:27:53 - INFO - No video currently playing, playing url : https://www.youtube.com/watch?v=Hrv25UTR_co
06-06 11:28:07 - INFO - Extracting source video URL...
06-06 11:28:07 - DEBUG - Parsing source url for https://www.youtube.com/watch?v=Hrv25UTR_co with subs :False
06-06 11:28:08 - DEBUG - [youtube] Hrv25UTR_co: Downloading webpage
06-06 11:28:12 - DEBUG - [youtube] Hrv25UTR_co: Downloading video info webpage
06-06 11:28:13 - DEBUG - [youtube] Hrv25UTR_co: Extracting video information
06-06 11:28:13 - DEBUG - [youtube] Hrv25UTR_co: Downloading MPD manifest
06-06 11:28:20 - DEBUG - CASTING: Youtube link detected, extracting url in maximal quality.
06-06 11:28:20 - DEBUG - Full video URL fetched.
06-06 11:28:20 - INFO - Starting OMXPlayer now.
06-06 11:28:51 - INFO - Playlist empty, skipping.

vincelwt, 2016-06-07:
Does it works with most of other videos ?
Or no Youtube videos at all works ?

boskodin, 2016-06-08:
It's not a severe issue, I just wanted to point out the inconvenience. I'm new to the rpi and github. 
I've come across at least 20 youtube videos that didn't cast. Others work fine.

Regards