vincelwt/RaspberryCast #21: Installing error

EricHier, 2016-06-11:
Hello, 

if I try to install the RaspberryCast after the reboot nothing happend. The log before the reebot: 

------------------------------------------------------------------------------

pi@raspberrypi:~ $ wget https://raw.githubusercontent.com/vincelwt/RaspberryCast/master/setup.sh && sudo sh setup.sh
converted 'https://raw.githubusercontent.com/vincelwt/RaspberryCast/master/setup.sh' (ANSI_X3.4-1968) -> 'https://raw.githubusercontent.com/vincelwt/RaspberryCast/master/setup.sh' (UTF-8)
--2016-06-11 11:58:29--  https://raw.githubusercontent.com/vincelwt/RaspberryCast/master/setup.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 185.31.17.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.31.17.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 2308 (2.3K) [text/plain]
Saving to: 'setup.sh'

setup.sh            100%[=====================>]   2.25K  --.-KB/s   in 0.001s

2016-06-11 11:58:30 (2.42 MB/s) - 'setup.sh' saved [2308/2308]

This script will install RaspberryCast
Do you wish to continue? (y/n)
y

============================================================

Installing necessary dependencies... (This could take a while)

============================================================
Reading package lists... Done
Building dependency tree
Reading state information... Done
libnss-mdns is already the newest version.
libnss-mdns set to manually installed.
wget is already the newest version.
The following extra packages will be installed:
  fontconfig-config fonts-dejavu-core fonts-droid fonts-freefont-ttf
  ghostscript git-man gsfonts libavahi-client3 libcups2 libcupsfilters1
  libcupsimage2 libcurl3-gnutls liberror-perl libexif12 libfontconfig1 libgif4
  libgs9 libgs9-common libijs-0.35 libjasper1 libjbig0 libjbig2dec0 liblcms2-2
  libldb1 libntdb1 libpaper-utils libpaper1 libperl4-corelibs-perl
  libsmbclient libssh-4 libtdb1 libtevent0 libtiff5 poppler-data
  python-chardet python-colorama python-distlib python-html5lib
  python-ndg-httpsclient python-openssl python-pkg-resources python-pyasn1
  python-requests python-setuptools python-six python-talloc python-urllib3
  python-wheel rsync samba-libs
Suggested packages:
  imagemagick ghostscript-x gettext-base git-daemon-run git-daemon-sysvinit
  git-doc git-el git-email git-gui gitk gitweb git-arch git-cvs git-mediawiki
  git-svn cups-common libjasper-runtime liblcms2-utils x11-xserver-utils
  poppler-utils fonts-japanese-mincho fonts-ipafont-mincho
  fonts-japanese-gothic fonts-ipafont-gothic fonts-arphic-ukai
  fonts-arphic-uming fonts-nanum python-genshi python-lxml python-openssl-doc
  python-openssl-dbg python-distribute python-distribute-doc doc-base
Recommended packages:
  python-dev-all
The following NEW packages will be installed:
  fbi fontconfig-config fonts-dejavu-core fonts-droid fonts-freefont-ttf
  ghostscript git git-man gsfonts libavahi-client3 libcups2 libcupsfilters1
  libcupsimage2 libcurl3-gnutls liberror-perl libexif12 libfontconfig1 libgif4
  libgs9 libgs9-common libijs-0.35 libjasper1 libjbig0 libjbig2dec0 liblcms2-2
  libldb1 libntdb1 libpaper-utils libpaper1 libperl4-corelibs-perl
  libsmbclient libssh-4 libtdb1 libtevent0 libtiff5 lsof omxplayer
  poppler-data python-chardet python-colorama python-distlib python-html5lib
  python-ndg-httpsclient python-openssl python-pip python-pkg-resources
  python-pyasn1 python-requests python-setuptools python-six python-talloc
  python-urllib3 python-wheel rsync samba-libs
0 upgraded, 55 newly installed, 0 to remove and 0 not upgraded.
Need to get 4379 kB/34.5 MB of archives.
After this operation, 123 MB of additional disk space will be used.
Err http://mirrordirector.raspbian.org/raspbian/ jessie/main samba-libs armhf 2:4.2.10+dfsg-0+deb8u2
  404  Not Found [IP: 2001:41c9:1:3ce::11 80]
Err http://mirrordirector.raspbian.org/raspbian/ jessie/main libsmbclient armhf 2:4.2.10+dfsg-0+deb8u2
  404  Not Found [IP: 2001:41c9:1:3ce::11 80]
E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/s/samba/samba-libs_4.2.10+dfsg-0+deb8u2_armhf.deb  404  Not Found [IP: 2001:41c9:1:3ce::11 80]

E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/s/samba/libsmbclient_4.2.10+dfsg-0+deb8u2_armhf.deb  404  Not Found [IP: 2001:41c9:1:3ce::11 80]

E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
============================================================
setup.sh: 37: setup.sh: pip: not found

============================================================

Cloning project from GitHub..

============================================================
-su: git: command not found
chmod: cannot access './RaspberryCast/RaspberryCast.sh': No such file or directory

============================================================

Adding project to startup sequence and custom options

============================================================
============================================================
Setup was successful.
Do not delete the 'RaspberryCast' folder as it contains all application data!
Rebooting system now...
============================================================

Broadcast message from pi@raspberrypi on pts/0 (Sat 2016-06-11 11:58:48 CEST):

The system is going down for reboot NOW!

------------------------------------------------------------------------------

Yes, thats it. I hope you can help me. 

Eric :)

vincelwt, 2016-06-11:
Hello,

That probably means your apt-get repos are outdated.

You may try to run : 

`sudo apt-get update`

Then download and relaunch the install script but without the rc.local part :

```
wget https://raw.githubusercontent.com/vincelwt/RaspberryCast/master/setup.sh 
sed '64,70d' setup.sh
sudo sh setup.sh
```

It should now install normally !

slimboyfats, 2016-07-03:
I get the same error, i'm on OSMC could that be the problem?

vincelwt, 2016-07-04:
@slimboyfats RaspberryCast isn't designed at all to be run on OSMC, but if you have the exact same error, then my previous answer should fix it.

mheine, 2018-07-04:
Closing issue due to old age and inactivity - feel free to re-open if the problem arises again 🙂

closed 2018-07-04 by mheine