vincelwt/RaspberryCast #22: Youtube Cast didn't work for me

hamdipro, 2016-08-15:
i installed RasberryCast for RasperryPi2 and cast feature didn't work either from web or mobile remote interface.
and i get these logs :

`08-15 16:44:00 - DEBUG - Received URL to cast: https://www.youtube.com/watch?v=IIJ30SVJwuQ
08-15 16:44:00 - DEBUG - No subtitles for this stream
08-15 16:44:00 - INFO - Extracting source video URL...
08-15 16:44:00 - DEBUG - Parsing source url for https://www.youtube.com/watch?v=IIJ30SVJwuQ with subs :False
08-15 16:44:00 - ERROR - ERROR: no suitable InfoExtractor for URL https://www.youtube.com/watch?v=IIJ30SVJwuQ
08-15 16:44:00 - ERROR - Result is none, returning none. Cancelling following function.
08-15 16:44:00 - DEBUG - Full video URL fetched.
08-15 16:44:00 - INFO - Starting OMXPlayer now.
08-15 16:44:00 - INFO - Playlist empty, skipping.
`

vincelwt, 2016-08-15:
Hi,

Could you try with another video? Like this one : https://www.youtube.com/watch?v=p3q5zWCw8J4
Also, could you try replacing https with http?

The error seems to come from youtube-dl, although I don't know why it doesn't works. 

Also you can try to reinstall youtube-dl on your pi:

```
sudo pip uninstall youtube-dl
sudo pip install youtube-dl
```

hamdipro, 2016-08-15:
Thank you for your response vincelwt
i fixed this issue by reinstalling youtube-dl but i get another issue.

`08-15 23:24:13 - DEBUG - Remote page requested.
08-15 23:24:17 - DEBUG - Received URL to cast: https://www.youtube.com/watch?v=aMIESSL6W60
08-15 23:24:17 - DEBUG - No subtitles for this stream
08-15 23:24:17 - INFO - Extracting source video URL...
08-15 23:24:17 - DEBUG - Parsing source url for https://www.youtube.com/watch?v=aMIESSL6W60 with subs :False
08-15 23:24:17 - DEBUG - [youtube] aMIESSL6W60: Downloading webpage
08-15 23:24:18 - DEBUG - [youtube] aMIESSL6W60: Downloading video info webpage
08-15 23:24:18 - DEBUG - [youtube] aMIESSL6W60: Extracting video information
08-15 23:24:19 - DEBUG - [youtube] aMIESSL6W60: Downloading js player en_US-vflduS31F
08-15 23:24:21 - WARNING - "id" field is not a string - forcing string conversion
08-15 23:24:22 - DEBUG - CASTING: Youtube link detected, extracting url in maximal quality.
08-15 23:24:22 - ERROR - Error in launchvideo function or during downlading the subtitles
08-15 23:24:22 - ERROR - 'url'
Traceback (most recent call last):
  File "./server.py", line 79, in stream
    launchvideo(url, False)
  File "/home/pi/RaspberryCast/process.py", line 15, in launchvideo
    out = return_full_url(url, sub)
  File "/home/pi/RaspberryCast/process.py", line 71, in return_full_url
    return video['url']
KeyError: 'url'
`