vincelwt/RaspberryCast #23: Playing music on two devices

warflash, 2016-09-05:
I was wondering if you could implement a feature that starts the video on your pc to the same time as it starts on the Raspberry, i think that would be handy when listening to music, thanks alot, love the project! ;)

vincelwt, 2016-09-05:
Hi @warflash 

The feature you're describing would be very hard to implement, mostly reducing the latency so you can have both the PC and Rpi playing the exact same thing at the exact same time.

If someone however is willing to spend some time trying to implement this, I can provide some help

dbazim, 2017-03-25:
Hello, for me playing sound simultaneously on phone/PC and RaspPi looks nor realistic case. But I really can't submit *.mp3 from remote device and use "cust now" That looks easier then playing video. Add this feature please. Audio Play list upload from Android/PC looks like useful feature as well.

Thanks in advance