vincelwt/RaspberryCast #24: Local video files on MacOSX?

TomZandt, 2016-09-24:
Hi,

I noticed you had support for local files on Linux. Any idea if its possible to do the same on MacOSX, either as a automater app, through terminal or in browser drag and drop?

Would love to be able to send files from my desktop and play them on the raspberry pi.

Thanks

vincelwt, 2016-10-09:
A while ago I made a GTK based Python script for all platforms, but it needs a lot of dependencies.
I'll just rewrite it for Command line so it just needs python, or maybe even in bash so there is no need for dependencies

TomZandt, 2016-10-09:
That would be amazing, thanks!

vincelwt, 2016-10-09:
Alright just finished it:
https://github.com/vincelwt/RaspberryCast#cast-videos-from-computer

It works with most of my videos, but I didn't tried with a lot

TomZandt, 2016-10-11:
Seems to work well, thanks.

I have been trying with h264 .mp4 files and works well. 

I do have a problem where if i ctrl+c out of the video (in terminal), as the verbose states, i can no longer play another file without a reboot of the pi?

The other thing is sometimes a get an error 400 and just wondered what that meant?

Ill have to have a look at the code when im less busy and see if i can understand how it works! 

For the time being its just what i was after.

Thanks

vincelwt, 2016-10-12:
Mmh, could you try first quiting the video from a RaspberryCast remote (eg Chrome extension) and then Ctrl+C ? 

Abou the 400, I guess it means than a request has been broken, or is incorrect.
Is it happening with the same video ? Does the Raspberry Pi and computer have a good internet connection ?

closed 2019-01-19 by vincelwt