vincelwt/RaspberryCast #32: Cannot exit from splash screen

griguolcomerranas, 2017-02-19:
Hi,

After the boot process I see briefly my Raspbian Desktop and then I get the splash screen "RaspberryCast is ready". In this state I can cast videos and they work great, but after a video is finished it gets back to the "RaspberryCast is ready" screen. I would like to go back to my Raspbian Desktop instead.

Is there any way to start RaspberryCast server in the background so it still listens to cast requests but I can meanwhile use my Raspbian Desktop?

Thanks!

vincelwt, 2017-02-20:
Hey,
you can disable the splash screen by changing the "_new_log_" option in the _raspberrycast.conf_ file by _false_

griguolcomerranas, 2017-02-20:
Wow. This is great! Now it is working exactly as I need it.  :)

I think this should be explained in front page README, I couldn't find it anywhere.

Thanks for your help!

closed 2017-02-20 by griguolcomerranas