vincelwt/RaspberryCast #34: Exit After Playback

Alladinsane, 2017-03-09:
I guess this may be more considered a feature request, but I would like RaspberryCast to "exit" after playback finishes.
To give my own usecase as an example: I have a setup where most of the time my Pi is displaying "digital signage", which is essentially just Chromium running in fullscreen kiosk mode. If I trigger omxplayer from command line, the video plays fullscreen, and when it is done the player closes, and my pi returns to displaying my digital sign. 
Currently, when I use RaspberryCast to send a video to my Pi, it correctly plays fullscreen in front of Chromium, but when it finishes, instead of closing out so that the digital sign is displayed again, it displays the RaspberryCast ready screen.

vincelwt, 2017-03-09:
Hey,
you can disable the splash screen by changing the "new_log" option in the raspberrycast.conf file by false, then reboot.

Alladinsane, 2017-03-09:
Awesome! Thanks!

closed 2017-03-09 by Alladinsane