vincelwt/RaspberryCast #36: Can't install RaspberryCast

rinlv, 2017-04-28:
I has been followed guide to install RaspberryCast 3.0, running this command:

`wget https://raw.githubusercontent.com/vincelwt/RaspberryCast/master/setup.sh && sudo sh setup.sh`

in the terminal, but always fail. I wonder, this script depend on OS, cause i'm using Raspbian.

vincelwt, 2017-04-28:
Hi,
Please send the error showing when the script fail.

rinlv, 2017-04-30:
Hi,
I was reinstalled OS and RaspberryCast successful. But when i want to cast 1 video from youtube, nothing change. Screen always display like this:

[https://goo.gl/photos/1xmUAFBFzzZ7rCqN9](url)

I'm wrong when configuration, aren't I?

vincelwt, 2017-05-05:
Could you provide the log, located at /home/pi/RaspberryCast/RaspberryCast.log ?

rinlv, 2017-05-07:
This is my RaspberryCast.log:
https://drive.google.com/open?id=0B3a_3TGa7ic8SmRHVWpaZDAzUVE

Thanks.