vincelwt/RaspberryCast #38: Unable to play any kind of content

nikolafilipovic, 2017-07-28:
Hello everyone,

I've come across a weird issue, when I submit a YouTube URL or any kind of URL as a matter of fact, the splash screen changes from "is ready" to "processing".

However, it never plays a clip, just goes back to "is ready". Here's the log...

`07-28 13:48:11 - DEBUG - Received URL to cast: https://www.youtube.com/watch?v=TUC2b-OSZ00
07-28 13:48:11 - DEBUG - No subtitles for this stream
07-28 13:48:11 - INFO - Extracting source video URL...
07-28 13:48:11 - DEBUG - Parsing source url for https://www.youtube.com/watch?v=TUC2b-OSZ00 with subs :False
07-28 13:48:11 - WARNING - Assuming --restrict-filenames since file system encoding cannot encode all characters. Set the LC_ALL environment variable to fix this.
07-28 13:48:12 - DEBUG - [youtube:playlist] playlist Serviio: Downloading 107 videos
07-28 13:48:12 - DEBUG - [download] Downloading video 1 of 107
07-28 13:48:12 - DEBUG - [download] Downloading video 2 of 107
07-28 13:48:12 - DEBUG - [download] Downloading video 3 of 107
07-28 13:48:12 - DEBUG - [download] Downloading video 4 of 107
07-28 13:48:12 - DEBUG - [download] Downloading video 5 of 107
07-28 13:48:12 - DEBUG - [download] Downloading video 6 of 107
07-28 13:48:12 - DEBUG - [download] Downloading video 7 of 107
07-28 13:48:12 - DEBUG - [download] Downloading video 8 of 107
07-28 13:48:12 - DEBUG - [download] Downloading video 9 of 107
07-28 13:48:12 - DEBUG - [download] Downloading video 10 of 107
07-28 13:48:12 - DEBUG - [download] Downloading video 11 of 107
07-28 13:48:12 - DEBUG - [download] Downloading video 12 of 107
07-28 13:48:12 - DEBUG - [download] Downloading video 13 of 107
07-28 13:48:12 - DEBUG - [download] Downloading video 14 of 107
07-28 13:48:12 - DEBUG - [download] Downloading video 15 of 107
07-28 13:48:12 - DEBUG - [download] Downloading video 16 of 107
07-28 13:48:12 - DEBUG - [download] Downloading video 17 of 107
07-28 13:48:12 - DEBUG - [download] Downloading video 18 of 107
07-28 13:48:12 - DEBUG - [download] Downloading video 19 of 107
07-28 13:48:12 - DEBUG - [download] Downloading video 20 of 107
07-28 13:48:12 - DEBUG - [download] Downloading video 21 of 107
07-28 13:48:12 - DEBUG - [download] Downloading video 22 of 107
07-28 13:48:12 - DEBUG - [download] Downloading video 23 of 107
07-28 13:48:12 - DEBUG - [download] Downloading video 24 of 107
07-28 13:48:12 - DEBUG - [download] Downloading video 25 of 107
07-28 13:48:12 - DEBUG - [download] Downloading video 26 of 107
07-28 13:48:12 - DEBUG - [download] Downloading video 27 of 107
07-28 13:48:12 - DEBUG - [download] Downloading video 28 of 107
07-28 13:48:12 - DEBUG - [download] Downloading video 29 of 107
07-28 13:48:12 - DEBUG - [download] Downloading video 30 of 107
07-28 13:48:12 - DEBUG - [download] Downloading video 31 of 107
07-28 13:48:12 - DEBUG - [download] Downloading video 32 of 107
07-28 13:48:12 - DEBUG - [download] Downloading video 33 of 107
07-28 13:48:12 - DEBUG - [download] Downloading video 34 of 107
07-28 13:48:12 - DEBUG - [download] Downloading video 35 of 107
07-28 13:48:12 - DEBUG - [download] Downloading video 36 of 107
07-28 13:48:12 - DEBUG - [download] Downloading video 37 of 107
07-28 13:48:13 - DEBUG - [download] Downloading video 38 of 107
07-28 13:48:13 - DEBUG - [download] Downloading video 39 of 107
07-28 13:48:13 - DEBUG - [download] Downloading video 40 of 107
07-28 13:48:13 - DEBUG - [download] Downloading video 41 of 107
07-28 13:48:13 - DEBUG - [download] Downloading video 42 of 107
07-28 13:48:13 - DEBUG - [download] Downloading video 43 of 107
07-28 13:48:13 - DEBUG - [download] Downloading video 44 of 107
07-28 13:48:13 - DEBUG - [download] Downloading video 45 of 107
07-28 13:48:13 - DEBUG - [download] Downloading video 46 of 107
07-28 13:48:13 - DEBUG - [download] Downloading video 47 of 107
07-28 13:48:13 - DEBUG - [download] Downloading video 48 of 107
07-28 13:48:13 - DEBUG - [download] Downloading video 49 of 107
07-28 13:48:13 - DEBUG - [download] Downloading video 50 of 107
07-28 13:48:13 - DEBUG - [download] Downloading video 51 of 107
07-28 13:48:13 - DEBUG - [download] Downloading video 52 of 107
07-28 13:48:13 - DEBUG - [download] Downloading video 53 of 107
07-28 13:48:13 - DEBUG - [download] Downloading video 54 of 107
07-28 13:48:13 - DEBUG - [download] Downloading video 55 of 107
07-28 13:48:13 - DEBUG - [download] Downloading video 56 of 107
07-28 13:48:13 - DEBUG - [download] Downloading video 57 of 107
07-28 13:48:13 - DEBUG - [download] Downloading video 58 of 107
[... 98 lines cut ...]
07-28 13:53:09 - DEBUG - Received URL to cast: http://techslides.com/demos/sample-videos/small.webm
07-28 13:53:09 - DEBUG - No subtitles for this stream
07-28 13:53:09 - INFO - Extracting source video URL...
07-28 13:53:09 - DEBUG - Parsing source url for http://techslides.com/demos/sample-videos/small.webm with subs :False
07-28 13:53:09 - WARNING - Assuming --restrict-filenames since file system encoding cannot encode all characters. Set the LC_ALL environment variable to fix this.
07-28 13:53:15 - DEBUG - [generic] small: Requesting header
07-28 13:53:22 - WARNING - "url" field is missing or empty - skipping format, there is an error in extractor
07-28 13:53:22 - ERROR - ERROR: list index out of range
07-28 13:53:22 - ERROR - Result is none, returning none. Cancelling following function.
07-28 13:53:22 - DEBUG - Full video URL fetched.
07-28 13:53:22 - INFO - Starting OMXPlayer now.
07-28 13:53:22 - INFO - Playlist empty, skipping.
07-28 14:01:03 - DEBUG - Received URL to cast: http://techslides.com/demos/sample-videos/small.mp4
07-28 14:01:03 - DEBUG - No subtitles for this stream
07-28 14:01:03 - INFO - Extracting source video URL...
07-28 14:01:03 - DEBUG - Parsing source url for http://techslides.com/demos/sample-videos/small.mp4 with subs :False
07-28 14:01:03 - DEBUG - Direct video URL, no need to use youtube-dl.
07-28 14:01:03 - DEBUG - Full video URL fetched.
07-28 14:01:03 - INFO - Starting OMXPlayer now.
07-28 14:01:04 - INFO - Playlist empty, skipping.
`

Not sure what's happening, this is on a Raspberry Pi 1 and I have it connected to an HDMI Monitor.

Any help would be greatly appreciated.

vmRelaxed, 2018-02-25:
07-28 14:01:04 - INFO - Playlist empty, skipping.
That explains something.
It cannot fetch it it seems like.
For me i waited like 10 / 20 mins to get this soundcloud playlist
https://soundcloud.com/jasonlaville/sets/best-deep-house-chill-music
Try setting 
07-28 13:48:49 - WARNING - Assuming --restrict-filenames since file system encoding cannot encode all characters. Set the LC_ALL environment variable to fix this.
Search on google how to set envvars and what they are.
07-28 13:49:09 - WARNING - "id" field is not a string - forcing string conversion, there is an error in extractor
Some problem to your link,

`
RaspberryCast - INFO - Server successfully started!
RaspberryCast - INFO - Processing playlist.
RaspberryCast - INFO - Playing first video of playlist
using "Liberation Mono-16", pixelsize=21.33 file=/usr/share/fonts/truetype/liberation2/LiberationMono-Regular.ttf
RaspberryCast - INFO - Extracting source video URL...
RaspberryCast - INFO - Starting OMXPlayer now.
RaspberryCast - INFO - Attempting to read resolution from configuration file.
RaspberryCast - INFO - Adding every videos from playlist to queue.
RaspberryCast - INFO - queuing video
RaspberryCast - INFO - queuing video
RaspberryCast - INFO - Extracting source video URL, before adding to queue...
Audio codec mp3 channels 2 samplerate 44100 bitspersample 16
Subtitle count: 0, state: off, index: 1, delay: 0
RaspberryCast - INFO - Full video URL fetched.
RaspberryCast - INFO - queuing video
RaspberryCast - INFO - Extracting source video URL, before adding to queue...
RaspberryCast - INFO - Full video URL fetched.
RaspberryCast - INFO - queuing video
RaspberryCast - INFO - Extracting source video URL, before adding to queue...
RaspberryCast - INFO - Full video URL fetched.
RaspberryCast - INFO - queuing video
RaspberryCast - INFO - Extracting source video URL, before adding to queue...
RaspberryCast - INFO - Full video URL fetched.
RaspberryCast - INFO - queuing video
RaspberryCast - INFO - Extracting source video URL, before adding to queue...
RaspberryCast - INFO - Full video URL fetched.
RaspberryCast - INFO - queuing video
RaspberryCast - INFO - Extracting source video URL, before adding to queue...
RaspberryCast - INFO - Full video URL fetched.
RaspberryCast - INFO - queuing video
RaspberryCast - INFO - Extracting source video URL, before adding to queue...
RaspberryCast - INFO - Full video URL fetched.
RaspberryCast - INFO - queuing video
RaspberryCast - INFO - Extracting source video URL, before adding to queue...
RaspberryCast - INFO - Full video URL fetched.
RaspberryCast - INFO - queuing video
RaspberryCast - INFO - Extracting source video URL, before adding to queue...
RaspberryCast - INFO - Full video URL fetched.
RaspberryCast - INFO - queuing video
RaspberryCast - INFO - Extracting source video URL, before adding to queue...
RaspberryCast - INFO - Full video URL fetched.
RaspberryCast - INFO - queuing video
RaspberryCast - INFO - Extracting source video URL, before adding to queue...
RaspberryCast - INFO - Full video URL fetched.
RaspberryCast - INFO - queuing video
RaspberryCast - INFO - Extracting source video URL, before adding to queue...
RaspberryCast - INFO - Full video URL fetched.
RaspberryCast - INFO - queuing video
RaspberryCast - INFO - Extracting source video URL, before adding to queue...
have a nice day ;)
RaspberryCast - INFO - Starting next video in playlist.
RaspberryCast - INFO - Starting OMXPlayer now.
RaspberryCast - INFO - Attempting to read resolution from configuration file.
Audio codec mp3 channels 2 samplerate 44100 bitspersample 16
Subtitle count: 0, state: off, index: 1, delay: 0
RaspberryCast - INFO - Full video URL fetched.
RaspberryCast - INFO - queuing video
RaspberryCast - INFO - Extracting source video URL, before adding to queue...
`
My log.

mheine, 2018-02-25:
I'm not quite sure if I understand your issue @vmRelaxed , but from what I understand, it can be fixed by updating youtube-dl. On your RaspberryPi, run 
`sudo pip install -U youtube-dl`

I am able to play both individual tracks and playlists with the updated version of youtube-dl. A possibility for permanently fixing these sort of errors is to always check for an updated versions of the python packages we're using on startup.

DanielGlaas, 2020-01-02:
Is there still something to do for this issue?