vincelwt/RaspberryCast #41: RetroPie/EmulationStation handoff after RaspberryCast done streaming

MrSco, 2017-10-10:
Great project! thanks for sharing it!...

Any tips for allowing emulationstation to take control of the system over after RaspberryCast has stopped streaming? I'm using RetroPie and RaspberryCast together and everything works pretty well except RaspberryCast doesn't give control of the screen back to emulationstation so a reboot is needed in order to start playing games again. emulationstation does this hand off back and forth between retroarch and itself all the time so there should be an easy way to accomplish this.

thanks again!

mheine, 2017-11-21:
I might give this a go, is the issue still relevant @MrSco ? Haven't used RetroPie, any more info you could give me?

MrSco, 2017-11-21:
yes, very relavant. i'm still having issues getting emulationstation back to the foreground after using raspberrycast to play a video. the fbi image gets thrown up after stopping the video and takes over the screen. the controls to emulationstation still work in the background so if you select a game and launch it, it will push the raspberrycast image to the background.... but even then, it still ends up being in the foreground on top of other shell type screens like the retropie setup blue screens, etc.

i'll try and help the best i can.... python and linux are still new to me.

thanks!

mheine, 2017-11-30:
I haven't been able to make any major progress on this. The case is very user-specific, so I think you'd be better off modifying your own version of RaspberryCast. You could try changing _server.py_ around lines 131 by adding a special case when "stop" is called:
`
elif control == "stop":
		logger.info('Command : stop video')
		os.system("cd <path_to_emulationstation> && ./emulationstation")
		return "1"
`

MrSco, 2017-11-30:
@mheine I actually had already tried exactly that... but it just launches an additional instance of emulationstation on top of raspberrycast (which is on top of the original instance of emulationstation). There must be a command that can just bring the original instance back to the foreground. The only workaround I came up with is to do "killall emulationstation" first and then "./emulationstation" but I don't think that shuts emulationstation down gracefully and also takes 30 seconds or so to load back up...

mheine, 2017-11-30:
Perhaps something along the lines of: `pkill -STOP emulstation`?

MrSco, 2017-11-30:
@mheine  to shutdown emulationstation gracefully? yea that may work, but i know there has to be a more efficient way to reuse the existing emulationstation process... its still there working, responding to the controller and you can even launch a game if you can blindly select one... and then the emulator/game loads and takes the foreground on top of raspberrycast... and upon quitting the game it returns you to emulationstation which is now also in the foreground on top of raspberrycast... so whatever command or steps that happen when you launch/quit a game from emulationstation is what needs to be recreated/simulated on raspberrycast video stop. 
.

mheine, 2017-11-30:
Aplogies, should say `pkill -CONT emulstation`.

MrSco, 2017-11-30:
@mheine  no go.. that doesn't seem to do anything. i'm guessing since the emulationstation process was never paused... so could i pause it first on raspberrycast play ... and then the pkill -CONT will unpause in foreground?

MrSco, 2017-11-30:
@mheine also killing emulationstation and relaunching seems to break launching games after that... gotta reboot to fix.

MrSco, 2017-11-30:
> However, not every user has Emulationstation installed, and as such it would be weird to always simulate it on RaspberryCast. There are many types of emulators, and making cases for each one seems weird. Hopefully we can solve your problem, but this doesn't feel like a global issue to me.

@mheine  I completely agree. I'm fine with just modding my server.py to accomodate my needs... just wish i could figure out the magic bullet to make this work silky smooth... so close!

madmodder123, 2018-02-24:
@MrSco Did you ever find a fix for this? I am running into the same issue and looking for a solution.

MrSco, 2018-02-24:
No, unfortunately I was never able to figure this out. Once raspberrycast takes over the screen with a video, it stays on top of emulationstation. It’s seems to be that way from raspberrycast getting launched before emulationstation on startup which puts it underneath but I’m not sure how to put it back underneath after a video is played

madmodder123, 2018-02-24:
@MrSco I posted about this issue on the RetroPie forums and someone found a solution: https://retropie.org.uk/forum/topic/16541/issues-re-focusing-emulationstation-raspberrycast/3

MrSco, 2018-02-24:
@madmodder123 yes! Thank you so much! Gonna Test it in a bit and I’ll report back.

madmodder123, 2018-03-05:
@MrSco 
I fixed the video issue...

I added "--aspect-mode fill"

So it looks like this now:

```
if sub:
    os.system(
        "omxplayer --aspect-mode fill -b --layer 10010 -o both '" + url + "' " + resolution + " --vol " +
        str(volume) +
        " --subtitles subtitle.srt < /tmp/cmd"
    )
elif url is None:
    pass
else:
    os.system(
        "omxplayer --aspect-mode fill -b --layer 10010 -o both '" + url + "' " + resolution + " --vol " +
        str(volume) + " < /tmp/cmd"
    )

```

MrSco, 2018-03-05:
Hmmm the video plays and looks ok. it still seems to black out whe hitting pause and play doesn’t start the video again.

mheine, 2018-06-20:
@MrSco and @madmodder123 , I'm going to close this issue, but feel free to re-open if you encounter similar problems down the line 🙂

closed 2018-06-20 by mheine