vincelwt/RaspberryCast #45: Firefox plugin should be adapted to WebExtension

yarons, 2017-11-27:
Firefox browser works only with WebExtension starting version 57 onward, any chance for a conversion?

https://wiki.mozilla.org/WebExtensions

vincelwt, 2017-11-29:
Yup! I don't really maintain this project anymore but if I start working on it again this'll be a priority.

yarons, 2017-11-30:
LOL thanks 😺

ardsur, 2018-09-21:
any update on possible upgrade for add-on FF Quatum (FF57+)?

mheine, 2018-09-22:
@ardsur I started working on it, and got it to a near-working state, but never finished for some reason. If you're interested in continuing the work (or if you have any experience with it), do tell me :slightly_smiling_face:

ardsur, 2018-09-23:
thanks Marcus for your answer.
sorry, but I do not have any programming skills so cannot help you out.
ard

yarons, 2018-09-23:
@mheine Where can I find your code?
Can you explain what you've done so far and what you're trying to achieve?

vincelwt, 2018-10-18:
Implement with #85

closed 2018-10-18 by vincelwt