vincelwt/RaspberryCast #49: RaspberryCast not working

Piguyy, 2018-01-14:
Hi!

I have installed RaspberryCast as it is stated in the guide and I am trying to make it run in the raspberry. I cannot make it to appear the splash image or anything. I have tried to run the RaspberryCast.sh file and it gives me the following error.

RaspberryCast.sh: 3: [: =: unexpected operator
RaspberryCast.sh: 17: [: =: unexpected operator
Error, wrong argument. Try with 'stop', 'start'.

I am not sure if I'm doing the correct thing or there is something that I am missing...

mheine, 2018-01-16:
As the last line of the error states, the script takes one argument, "start" or "stop". Try running:
`sudo ./Raspberrycast.sh start` :slightly_smiling_face:

Piguyy, 2018-01-16:
Thank you very much! It was that :)

closed 2018-01-16 by Piguyy