vincelwt/RaspberryCast #51: Open and run in background

octoshrimpy, 2018-01-18:
Would be lovely to have the option to have it run in the background. The "cast is ready" screen is constantly on top of all other windows, which makes it hard to see the dashboard I've setup for home automation..

audriusjanus, 2018-01-18:
Agree, this option would be amazing. I would to see my desktop to,  only when I cast  program would pop up player.

octoshrimpy, 2018-01-18:
@audriusjanus I found out that by doing this https://github.com/vincelwt/RaspberryCast/issues/48 you can actually turn off the "cast is ready/ cast is processing" screens, but then you don't know exactly if it's processing or if it broke or anything. having it be not there until you cast and then have the casting is processing screen, followed by the stream would be amazing.

mheine, 2018-01-19:
You could implement this yourself without too much hassle. Add a line in the configuration file (something like `"show_processing": true,`), then on line 18 in `process.py`you can substitute
`if config["new_log"]:`
and instead write:
`if config["show_processing"]:`

To me this feels like a very specific request, but I'd be happy to write a PR with this implemented.

vincelwt, 2018-02-12:
I agree, this is kind of a specific request. If more users feel like this option would be useful, this could be an easy add.

closed 2018-02-12 by vincelwt