vincelwt/RaspberryCast #56: I get SyntaxError: invalid syntax when I try to run Raspicast (the server.py script) on Arch Linux ARM

qwxy0, 2018-03-24:
I've been trying to install RaspberryCast on Arch Linux ARM manually. After following the install script and tracking down dependencies needed for RaspberryCast on Arch ARM and after cloning the git(hub) repo, when I try to start RaspberryCast I get the following error:

`[alarm@alarmpi RaspberryCast]$ ./RaspberryCast.sh start
Checking for updates.
Already up to date.
Starting RaspberryCast server.
Done.
[alarm@alarmpi RaspberryCast]$   File "./server.py", line 109
    except Exception, e:
                    ^
SyntaxError: invalid syntax`

Any insight into the error would be much appreciated. Could it be due to the version of python3 being too new or missing some module? I don't know really.
Thank you for your help

mheine, 2018-03-25:
I don't run Arch myself, but that does sound like a Python 2/3 version issue, and would probably be fixed by ensuring that you run it using Python 2 🙂

jarondl, 2018-06-05:
Yes, indeed, that is because of Python2/3 differences. RaspberryCast should definitely support python3, but for now you can work around this by reinstalling your dependencies with Python2 and running:

    python2 server.py &

PR #65 (2018-06-12): make server.py Python3 compatible
This makes `server.py` compatible with both Python2 and Python3.
The `expect as e` syntax works since Python 2.6, including all versions
of Python3. The urllib changed structure between Python2 and Python3

`process.py` was already 2/3 compatible. `rcast.py` needs a lot of work,
but can be done separately since it is run on another host.
`YoutubeFullUrl.py` is not used anywhere.

Fixes #56
commit 719fdf5d (2018-06-12): make server.py Python3 compatible

This makes `server.py` compatible with both Python2 and Python3.
The `expect as e` syntax works since Python 2.6, including all versions
of Python3. The urllib changed structure between Python2 and Python3

`process.py` was already 2/3 compatible. `rcast.py` needs a lot of work,
but can be done separately since it is run on another host.
`YoutubeFullUrl.py` is not used anywhere.

Fixes #56
closed 2018-06-12 by vincelwt