vincelwt/RaspberryCast #62: General questions and suggestions

doronbehar, 2018-05-20:
Hello @vincelwt ,

I'm very interested in using and contributing to this project. However, I think the way the server scripts are organized could use some improvements. I think this project could really shine if you'd organize it differently, here are a few suggestions:

- Why use a shell wrapper for the `server.py`? You can processes command line arguments inside python and I believe you can fork processes and make the same script fork it self and kill it self according to CLI arguments.
- The installation script is probably very NOOB friendly but it's not very distribution / system administrator friendly IMO. I use Arch Linux on my Raspberry Pi and I can't use this script because we don't use `/etc/rc.local`. We use strictly `systemd` services. I understand that `setup.sh` creates a user that the `server.py` will run under his privileges but IMHO it's not truly necessary. I think this should stay in the documentation only as a recommendation.
- In addition, as for using `/etc/rc.local` and especially putting stuff like [this](https://github.com/vincelwt/RaspberryCast/blob/2105100/setup.sh#L81) in `setup.sh`, it doesn't make it easy to uninstall RaspberryCast. With a `systemd` based configuration it will be much easier.
- If we'll get rid of the shell script wrapper and distribute only a python script that will manage it self, you can distribute a pip package that will have `youtube-dl` `bottle` and `livestreamer` in it's `requirements.txt` so they will be installed automatically.
- As for the rest of the dependencies (`lsof`, `python-pip`, `git`, `wget`, `omxplayer`, `libnss-mdns` and `fbi`) it could be even better to distribute a package that could be installed through https://launchpad.net/ (I can write a `PKGBUILD` for Arch Linux).

I'll be glad to know your thoughts :)

jarondl, 2018-06-05:
Hi @vincelwt  how open to changes are you? Since this is essentially a single-process app, I was thinking of converting the control mechanism from files on the filesystem to streams. Would you be accepting for big changes like this? I ask this here because it would make the distribution much easier if there are not files lying around the filesystem.

vincelwt, 2018-06-05:
Hi @jarondl , I'm open to improvements :) What files are you referring to transform to streams? RaspberryCast already stream videos. You can always fork the repo, implement what you want, and start a pull request if you feel it's ready for all RaspberryCast users.