vincelwt/RaspberryCast #69: [enhancement] Add streamlink as a alternative to youtube-dl

jiahaoliang, 2018-06-20:
[streamlink](https://github.com/streamlink/streamlink) is similar to youtube-dl in many different ways.

Although the video sources they supported are mostly the same, there are still some plugins available in streamlink that is missing in youtube-dl (live stream from douyu.com for example). Also it's always a good thing to add a duplicate extractor in case youtube-dl is broken in any way.

Interested in exploring how to add streamlink as a secondary extractor as well as a general framework to integrate other extractors in the future.

Hints and guidelines are always appreciated.

mheine, 2018-06-20:
Looks interesting! I've have some trouble with stuttering playback from livestreams on twitch.tv, so this might be another option. My suggestion would be to add a configuration option that sets the extractor to streamlink (but where the default would still be youtube-dl)

jarondl, 2018-06-20:
Seems like @vincelwt started something in the past (in  c4f76b9cc60bb63a1dc0cc5849f3c873b577b765  livestreamer is the predecessor of streamlink).

This seems like a great idea, but I would try to use their Python API instead of calling the cli.