vincelwt/RaspberryCast #71: [question] how can I view the videos in the playlist?

helen-fornazier, 2018-07-02:
Hi,

First of all, thanks for this project, is it great!

I added some videos from youtube to the playlist, it works, but is it possible to view the videos that are currently in the play list? Also, if is it possible to manipulate the playlist (delete some videos, clean the list)

Thanks

vincelwt, 2018-07-02:
Hi,
Unfortunately, this is not possible at the moment. 
There is already a feature request opened for it: https://github.com/vincelwt/RaspberryCast/issues/14
Cheers,
Vincent

closed 2018-07-02 by vincelwt

helen-fornazier, 2018-07-02:
Nice.

Sorry, I didn't see this feature request, thanks for pointing me to it.

mheine, 2018-07-03:
@helen-fornazier If you're interested in working on it, there's a (rather empty) branch called `manage-playlist ` that you can use as a starting point if you want :slightly_smiling_face:

helen-fornazier, 2018-07-03:
@mheine Cool, I'll take a look when I have so time, thanks