vincelwt/RaspberryCast #74: Unable to play from Voot

Vikassm73, 2018-07-25:
Youtube, local video works fine. but unable to play from Voot, Hotstar, zee5 etc..

Vikassm73, 2018-07-25:
Also, The raspberrycast cover the screen of raspbian. How to close this application so that I can use raspbian?

hungdq11, 2018-07-25:
Its depends on youtube-dl (https://github.com/rg3/youtube-dl)

hungdq11, 2018-07-25:
> Also, The raspberrycast cover the screen of raspbian. How to close this application so that I can use raspbian?
./RaspberryCast.sh stop

hungdq11, 2018-07-25:
Sorry, its also depends on omxplayer. In your case i saw youtube-dl returned HLS link so maybe it didn't work with omxplayer

Vikassm73, 2018-07-25:
I have already installed omxplayer. It works fine with raspiast as well as raspberrycast but only youtube.

mheine, 2018-07-25:
As mentioned above, if it cannot be downloaded by youtube-dl, it cannot be streamed by RaspberryCast. Also, if you want to use other programs and have RaspberryCast running in the background, set the `new_log` configuration option to `false` to not display the "RaspberryCast is ready" option 🙂

Vikassm73, 2018-07-27:
I have tried as it was mention in the supported site

http://rg3.github.io/youtube-dl/supportedsites.html