vincelwt/RaspberryCast #90: Play audio only

retorquere, 2019-02-24:
I'm looking for a way to cast videos (like youtube videos) to my receiver without it turning on my TV through CEC. This would allow me to play videos as if they were playing to a chromecast audio, it would urn on my receiver when I cast through CEC, but not turn the TV on (which is what a regular chromecast does).

jsfaure, 2019-04-07:
Hi @retorquere , I was looking for the same kind of feature. I gave it a try by redirecting the audio flow to the headphone jack output. This is far from being elegant in term of resources' usage but give a good result.
See https://www.raspberrypi.org/documentation/configuration/audio-config.md

retorquere, 2019-04-07:
Thanks. I'm not sure what I want to do is possible at all -- it looks like HDMI inherently sends audio/video at the same time (in the same packets); if that's so "audio-only HDMI" does not exist.

The reason why I wanted to do this via HDMI is that I want the cast device to turn on my AV receiver (which requires CEC) but then not have the TV that's also connected to the receiver turn on. From the looks of it, this will require getting an AV receiver where I can selectively disable CEC passthrough. Not even sure these exist. That, or an AV receiver that can detect activity on the optical or analog audio-in (also not sure whether these exist).

DanielGlaas, 2021-01-05:
Hey guys,
the omx player is automatically called with the parameter '-o both', which (at least for me) results into the effect that the video and sound are played via the TV if it is switched on and also via the headphone jack.
If both work in parallel I didn't test because this is not a use case for me.

DanielGlaas, 2021-01-06:
> Hey guys,
> the omx player is automatically called with the parameter '-o both', which (at least for me) results into the effect that the video and sound are played via the TV if it is switched on and [...]

Ok, I checked it. 
Output on the TV via HDMI and parallel output on the headphone jack works"