vincelwt/RaspberryCast #94: ERROR - ERROR

liakjim, 2019-04-26:
Hi, raspberry will not play youtube videos anymore. 
The screen says that video is processing, then ready but nothing happens.
RaspberryCast.log returns :
ERROR - ERROR: 4XDHmnzT46M: "token" parameter not in video info for unknown reason; please report this issue on https://yt-dl.org/bug . Make sure you are using the latest version; see  https://yt-dl.org/update  on how to update. Be sure to call youtube-dl with the --verbose flag and include its complete output.

I have latest version of youtube-dl installed. 
youtube-dl -U returns:
youtube-dl is up-to-date (2019.04.24)

tedchou12, 2019-05-02:
Seems like I am facing the same situation...

mheine, 2019-05-02:
Weird. Seems to be an issue with youtube-dl, and having a look at their issues there have been a few similar cases like this in the past, but they've all been attributed to old versions of youtube-dl.
Are you able to download the video "manually" (as in, using the standalone version of `youtube-dl`, and not using RaspberryCast)?

liakjim, 2019-05-02:
> Weird. Seems to be an issue with youtube-dl, and having a look at their issues there have been a few similar cases like this in the past, but they've all been attributed to old versions of [...]

It seems that it does : 
[image]

elblogbruno, 2019-05-04:
I fixed it by updating youtube-dl to the latest version and rebooting my raspberry pi!

liakjim, 2019-05-04:
> I fixed it by updating youtube-dl to the latest version and rebooting my raspberry pi!

Did you update with youtube-dl -U command? Or something else? If yes, which version it installed? In my case it returns this  : "youtube-dl is up-to-date (2019.04.24)".
The problem remains though.

elblogbruno, 2019-05-04:
I run `sudo -H pip install --upgrade youtube-dl`  to update using Pip.
Here's my youtube-dl version.
[image]
then run `sudo reboot` to reboot.

liakjim, 2019-05-04:
> I fixed it by updating youtube-dl to the latest version and rebooting my raspberry pi!

Did you update with youtube-dl -U command? Or something else? If yes, which version it installed? In my case it returns this  : "youtube-dl is up-to-date (2019.04.24)".
The problem remains though.

> I run `sudo -H pip install --upgrade youtube-dl` to update using Pip.
> Here's my youtube-dl version.
> [image]
> then run `sudo reboot` to reboot.

Oh, there is a new version out as it seems. I will try when back at home and report. Thanks

liakjim, 2019-05-04:
Ok, 2019.04.30 version seems to be working !!!!
Thanks @elblogbruno

elblogbruno, 2019-05-04:
@liakjim  You're welcome!

liakjim, 2019-05-04:
Now it seems to be another issue. 
When raspberry login, there is the "raspberrycast is ready" screen . 
https://i.postimg.cc/6QSk8V1C/ready.jpg 

It won't go away. 

@elblogbruno , is it the same with you?

elblogbruno, 2019-05-04:
I believe its not an error as RaspberryCast intention is to work that way, as a chromecast device when you plug it in a tv.
I disabled this screen because I didn't want to have it for my project, I'll show you how later if you need to.

liakjim, 2019-05-04:
> I believe its not an error as RaspberryCast intention is to work that way, as a chromecast device when you plug it in a tv.
> I disabled this screen because I didn't want to have it for my project, I'll show you how later if you need to.

Yes please.

elblogbruno, 2019-05-04:
Check my fork: https://github.com/elblogbruno/RaspberryCast
you'll see the commits i've made to the files

DanielGlaas, 2020-01-02:
Hi guys,
If you don't want that these screen are displayed, the just set the value of 'new_log' to 'false' in the raspberrycast.conf file.
This variable one serves for this purpose and  nothing else.

DanielGlaas, 2020-01-02:
As the requested behaviour can be archived by the configuration file, I recommend to close this issue.

elblogbruno, 2020-01-02:
Didn't know that thanks!

liakjim, 2020-01-09:
> 
> 
> Hi guys,
> If you don't want that these screens are displayed, then just set the value of 'new_log' to 'false' in the raspberrycast.conf file.
> This variable only serves for this purpose and for nothing else.

This fixed this issue. Many thanks!

closed 2020-01-09 by liakjim