vincelwt/RaspberryCast #95: Doesn't work on Google Chrome

REALSCHW4RZ, 2019-05-17:
The Chrome extension doesn't work for me. I get this error message all the time: "Error during accessing server. Make sure the ip/port are corrects, and the server is running."

I have tried adding the ports to the IP-Adress but it still won't work.

Best regards
Julian

omody, 2019-05-20:
This is correct.  I tried this myself and noticed the connection to host was being blocked because of CORS.  Apparently the hostname has to be provided in the "permissions" section of the extension.
The second thing was the jquery js file need to be first. Because the next error I saw was "$" is undefined.

REALSCHW4RZ, 2019-05-22:
So what do I have to do to fix it?

I don't know what you mean with "permissions" section...

omody, 2019-05-22:
I changed the manifest.json to below and it seems to be working now:

{
   "background": {
      "persistent": true,
      "scripts": [ "js/jquery-2.1.3.min.js", "js/background.js", "js/popup.js" ]
   },
   "browser_action": {
      "default_icon": "20.png",
      "default_title": "RasperryCast remote",
      "default_popup": "popup.html"
   },
   "description": "Share videos to Rpi",
   "icons": {
      "128": "128.png",
      "16": "16.png",
      "48": "48.png"
   },
   "options_page": "options.html",

   "manifest_version": 2,
   "name": "RaspberryCast",
   "permissions": [ "contextMenus", "tabs", "notifications", "http://raspberrypi.local/" ],
   "version": "1.3.0"
}

REALSCHW4RZ, 2019-05-23:
Ok, this still won't work for me. I don't understand because it works on my mobile phone....

aryamanr10, 2020-11-28:
This didn’t work for me either, now it will no longer connect to the server. From what I can tell there is no difference in between what is already in manifest.json and what ghost told me to change it to. The order of the scripts under background was changed but otherwise there was nothing.