vincelwt/RaspberryCast #101: Control controls do not work anymore on Raspbian Buster

twinstef, 2019-08-22:
Hello,
I try to use RaspberryCast on a Raspberry Pi3 A+ under Raspbian BUSTER
I can launch a video put can't control it.

- commands like pause, quit, volume ... etc do not works at all.

When i look to source code, the command are launched via the fifo file /tmp/cmd with code like: 
> echo -n q >> /tmp/cmd

But this way to pilot omxplayer does not work anymore on Buster !
See this issue: https://github.com/popcornmix/omxplayer/issues/737

Perhaps a more robust way to control omxplayer could be:
`p = subprocess.Popen(['/usr/bin/omxplayer ......'], stdin=subprocess.PIPE, stdout=subprocess.PIPE, stderr=subprocess.PIPE, shell=True)`
and then:
`p.communicate(input='q')`
But i not tried it to see if it works under buster

mheine, 2019-08-23:
There seems to be a `dbuscontrol.sh` script in omxplayer that (apparently?) works by sending dbus messages, which would be better than echoing commands to the fifo file. I am currently not able to implement or test this, but if anyone wants to give it a shot (either by using the script, or by using `dbus-send` straight up), feel free to submit a PR :)

twinstef, 2019-08-23:
I agree, i will test that when i will have some time.
I found a project based on dbus: 

> python-omxplayer-wrapper - https://github.com/willprice/python-omxplayer-wrapper

which can help to test or even implement a more robust way to communicate inside RaspberryCast

nemozar, 2020-02-06:
> I agree, i will test that when i will have some time.
> I found a project based on dbus:
> 
> > python-omxplayer-wrapper - https://github.com/willprice/python-omxplayer-wrapper
> 
> which can help to test or even implement a more robust way to communicate inside RaspberryCast

i have same problem in buster. At last version raspbian work correct? How you fix this probleb? I'm dosn't know python and not understand how it fix. Can you write instruction or create fork with fix this problem on buster?